exelban/stats #2431: Add TBW Value for SSDs in the Disk Section

emresaracoglu, 2025-03-05:
Hello,

This is one of the best tools I’ve used for Mac! Many thanks to everyone who contributed to its development. I have a request: If possible, could you add the TBW (Total Bytes Written) value for SSDs in the Disk section?

Thanks

exelban, 2025-03-05:
hi. Thanks.
There are already total write/read for disks.

emresaracoglu, 2025-03-05:
[image]

The ‘Total Write’ (on screenshot) value seems to reset when the computer is restarted. What I’m referring to is displaying the main TBW (\\_Data Units Written_) value permanently.

I can view this data using 'smartmontools' in the terminal, but I couldn't find it in the 'Stats' app. If it's already available, I'll check again; otherwise, it would be great to see it included in future versions.

``` 
emre@192 Desktop % sudo smartctl -a /dev/disk0 
...
Data Units Read:                    3 029 569 [1,55 TB]
Data Units Written:                 2 127 678 [1,08 TB] 
...
emre@192 Desktop % 
```

exelban, 2025-03-05:
I will take a look if these values could be obtained

emresaracoglu, 2025-03-05:
That would be great! If it's possible to include them in future versions, it would be very helpful. Thanks for looking into it!

closed 2025-03-16 by exelban

emresaracoglu, 2025-03-16:
Thank you for adding this feature

emresaracoglu, 2025-03-18:
[image]

exelban, 2025-03-18:
I will add a title that will show total units to these values, it will allow to figure out why the values are different

emresaracoglu, 2025-03-18:
This software is truly a masterpiece for me. I am sincerely grateful that you added the feature I requested; it means a lot to me. Thank you!  

By the way, I noticed something: The TBR & TBW values of external or iSCSI-connected HDD/SSD devices are not being detected. I conducted extensive research to access SMART data but couldn't find a solution. If I discover a method in the future, I would love to share it with you.

[image]

exelban, 2025-03-18:
Thanks)

Yeap, there are a lot of disks without SMART support (on hardware or software level). I'm not sure there is a workaround here(

checktext00, 2025-04-16:
> By the way, I noticed something: The TBR & TBW values of USB-A/C or iSCSI-connected HDD/SSD devices are not being detected. I conducted extensive research to access SMART data but couldn't find a solution. If I discover a method in the future, I would love to share it with you.

You need a special driver for macOS to read SMART data from a USB drive. You can download and install it as a Gatekeeper-signed .pkg from [https://binaryfruit.com/drivedx/usb-drive-support](https://binaryfruit.com/drivedx/usb-drive-support)

from the smartmontools wiki:
[https://www.smartmontools.org/wiki/USB](https://www.smartmontools.org/wiki/USB)

> The ​[OS X SAT SMART Driver](https://github.com/kasbert/OS-X-SAT-SMART-Driver) provides access to SMART data for SAT, JMicron and Sunplus capable USB and Firewire devices on Mac OS X. On OSX El Capitan (10.11+) you may need a signed driver, which can be downloaded from the ​[DriveDx website](http://binaryfruit.com/drivedx/usb-drive-support/). Note that this does not support 48-bit ATA commands (`-l xerror`) or non-SMART ATA commands (`-s apm,N`). Full USB support would require SCSI pass-through support which does not yet exist on Mac OS X.

You can check it's working if you have `smartctl` installed from the [smartmontools package](https://sourceforge.net/projects/smartmontools/files/smartmontools/) with:
```
smartctl -x /dev/disk2
```
(assuming your USB is disk2, check with `diskutil list`)