exelban/stats #2434: Stats app randomly turns off and re-opening application closes immediately

Geiko-jpg, 2025-03-09:
**Describe the bug**
Stats randomly turns off and cannot open the application

**Details:**
 - Device: [e.g. Macbook Pro M3 Pro]
 - macOS: [e.g. 15.3.1]
 - Application version: [e.g. 2.11.31]

exelban, 2025-03-09:
please share the crash report

gshen42, 2025-03-09:
Second this. Here's my crash report:
```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [5679]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.30 (673)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-09 10:03:48.3891 -0700
OS Version:            macOS 15.3.1 (24D70)
Report Version:        12
Anonymous UUID:        FC908188-D204-E2D7-7B28-814188E51170

Sleep/Wake UUID:       A393A752-C3B4-4B1D-B33C-63F127410A61

Time Awake Since Boot: 20000 seconds
Time Since Wake:       1729 seconds

System Integrity Protection: enabled

Crashed Thread:        2  Dispatch queue: com.apple.xpc.activity.eu.exelban.Stats.support

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000019afe63d8

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [5679]

Application Specific Information:
Must only be used from the main thread

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x196666f54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x196679604 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x19666faf8 mach_msg_overwrite + 480
3   libsystem_kernel.dylib        	       0x19666729c mach_msg + 24
4   CoreFoundation                	       0x196790a4c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x19678f2ac __CFRunLoopRun + 1212
6   CoreFoundation                	       0x19678e734 CFRunLoopRunSpecific + 588
7   HIToolbox                     	       0x1a1cfd530 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1a1d03348 ReceiveNextEventCommon + 676
9   HIToolbox                     	       0x1a1d03508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x19a306848 _DPSNextEvent + 660
11  AppKit                        	       0x19ac6cc24 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
12  AppKit                        	       0x19a2f9874 -[NSApplication run] + 480
13  Stats                         	       0x1041efa88 0x1041ec000 + 14984
14  dyld                          	       0x196328274 start + 2840

Thread 1:
0   libsystem_pthread.dylib       	       0x1966a30e8 start_wqthread + 0

Thread 2 Crashed::  Dispatch queue: com.apple.xpc.activity.eu.exelban.Stats.support
0   AppKit                        	       0x19afe63d8 -[NSWMWindowCoordinator performTransactionUsingBlock:] + 768
1   AppKit                        	       0x19ae8d258 _NSWMTransactionAddOrderingActionWithInfoForOptions + 184
2   AppKit                        	       0x19ae8c4cc _NSWMOrderWindow + 588
3   AppKit                        	       0x19a559088 _NSOrderWindow + 360
4   AppKit                        	       0x19ae08fc8 __45-[NSWindow _reallyDoOrderWindowAboveOrBelow:]_block_invoke + 452
5   AppKit                        	       0x19a3484d8 NSPerformVisuallyAtomicChange + 108
6   AppKit                        	       0x19ae08c4c -[NSWindow _reallyDoOrderWindowAboveOrBelow:] + 1464
7   AppKit                        	       0x19ae0966c -[NSWindow _reallyDoOrderWindow:] + 64
8   AppKit                        	       0x19ae098cc -[NSWindow _doOrderWindow:] + 264
9   Stats                         	       0x1042113bc 0x1041ec000 + 152508
10  Stats                         	       0x1042141c4 0x1041ec000 + 164292
11  Stats                         	       0x10420305c 0x1041ec000 + 94300
[... 607 lines cut ...]
    {
      "rolloutId" : "63f9578e238e7b23a1f3030a",
      "factorPackIds" : {

      },
      "deploymentId" : 240000005
    },
    {
      "rolloutId" : "661464ecda55e5192b100804",
      "factorPackIds" : {

      },
      "deploymentId" : 240000005
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "2093e844-0d7f-48e6-8857-c04fa52c3679",
      "experimentId" : "67be7deb213fff22c33c3e7e",
      "deploymentId" : 400000001
    }
  ]
}
}
```

closed 2025-03-09 by exelban