exelban/stats #2435: App doesn't work after waking up from sleeping

RyuunosukeH, 2025-03-09:
Nothing will be displayed sometimes after couple of sleeps. Tried to open the app when it happens but will instantly quit. Could only be resolved by restarting the system. Not sure what is the cause of the bug and how to solve it though.

exelban, 2025-03-09:
please share the crash report

veuda, 2025-03-10:
Hi,

Thanks for your work.

Same thing here, on Sequioa 15.3.1.

Here is the report :

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [7230]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.31 (674)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-10 14:33:36.0471 +0100
OS Version:            macOS 15.3.1 (24D70)
Report Version:        12
Anonymous UUID:        8C8B0B62-4346-73B7-3820-9C593D9227D2

Sleep/Wake UUID:       45B43787-1C07-40A3-87A5-D37D4083BADA

Time Awake Since Boot: 14000 seconds
Time Since Wake:       9133 seconds

System Integrity Protection: enabled

Crashed Thread:        7  Dispatch queue: com.apple.xpc.activity.eu.exelban.Stats.support

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000019c9f23d8

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [7230]

Application Specific Information:
Must only be used from the main thread

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x198072f54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x198085604 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x19807baf8 mach_msg_overwrite + 480
3   libsystem_kernel.dylib        	       0x19807329c mach_msg + 24
4   CoreFoundation                	       0x19819ca4c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x19819b2ac __CFRunLoopRun + 1212
6   CoreFoundation                	       0x19819a734 CFRunLoopRunSpecific + 588
7   HIToolbox                     	       0x1a3709530 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x1a370f348 ReceiveNextEventCommon + 676
9   HIToolbox                     	       0x1a370f508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x19bd12848 _DPSNextEvent + 660
11  AppKit                        	       0x19c678c24 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
12  AppKit                        	       0x19bd05874 -[NSApplication run] + 480
13  Stats                         	       0x100923914 0x100920000 + 14612
14  dyld                          	       0x197d34274 start + 2840

Thread 1:
0   libsystem_pthread.dylib       	       0x1980af0e8 start_wqthread + 0

Thread 2::  Dispatch queue: com.apple.root.background-qos
0   libsystem_kernel.dylib        	       0x198073ad0 read + 8
1   Foundation                    	       0x19933bdd8 _NSReadFromFileDescriptorWithProgress + 196
2   Foundation                    	       0x19935a818 -[NSConcreteFileHandle readDataOfLength:] + 380
3   RAM                           	       0x100aaf870 0x100aa4000 + 47216
4   Kit                           	       0x100bf5b08 0x100b78000 + 514824
[... 632 lines cut ...]
    },
    {
      "rolloutId" : "61675b89201f677a9a4cbd65",
      "factorPackIds" : {
        "HEALTH_FEATURE_AVAILABILITY" : "65a855f5f087695cfac03d1f"
      },
      "deploymentId" : 240000170
    }
  ],
  "experiments" : [
    {
      "treatmentId" : "f18344bb-554f-48ae-8449-e108a8813a55",
      "experimentId" : "6685b5584477001000e8c6c9",
      "deploymentId" : 400000004
    },
    {
      "treatmentId" : "28060e10-d4e5-4163-aa2b-d8bd088d5cbf",
      "experimentId" : "6685b283afc7c17197d69eec",
      "deploymentId" : 400000005
    }
  ]
}
}

```

exelban, 2025-03-10:
please read readme before posting. A few shared reports was not from the latest build. And the comments/issues like that. will be automatically deleted/closed.

BluGum, 2025-03-13:
Adding that I'm also experiencing this issue exactly as @Norbert-Mauger describes. Not sure how to get a crash report as Stats isn't open for long enough, it doesn't even have time to show up in Activity Monitor before it quits/crashes.

Device: M4 MacBook Pro
OS: macOS Sequoia 15.3.1

Attached is the dialog that displays for a fraction of a second in case it's helpful.

[image]

seahindeniz, 2025-03-13:
I was reproducing the same issue. After I download the latest version and did the installation from DMG file, the app started running again. I cannot reproduce anymore.

https://github.com/user-attachments/assets/12eec534-d4f9-45ad-b4b4-3f3de4b0ba99

exelban, 2025-03-13:
hi. I understand that you have a problem with running Stats. But based on just text "it does not work" or film/screenshot that I does not work I cannot help at all.

Please share a crash report, otherwise I cannot help

seahindeniz, 2025-03-13:
Ah, sorry about that @exelban. Here are the reports I have

[Archive.zip](https://github.com/user-attachments/files/19229087/Archive.zip)

closed 2025-03-13 by exelban

seahindeniz, 2025-03-13:
Yes, I have confirmed the latest version in my previous message 🙏🏻

exelban, 2025-03-13:
You confirm…
but crash report says you are wrong. Since the latest version is v2.11.34, and your is v2.11.31…

seahindeniz, 2025-03-13:
It was an old crash report from before I download the latest version. Since you still asked after the confirmation, I thought you wanted to debug it for further details. I have 2.11.34 when I confirmed it

exelban, 2025-03-13:
Please share the crash report of the latest build

seahindeniz, 2025-03-13:
There is non 🤔

exelban, 2025-03-13:
oh, you mean that after updating to the latest it was fixed. Sorry, misunderstand that

seahindeniz, 2025-03-13:
No worries, and actually, excuse my lack of explanation and missdirecting message. I should have been clear about it 🙏🏻 Thank you for your efforts