exelban/stats #2436: [REQUEST] For CPU warnings, include "more than x seconds" or something

dzg, 2025-03-10:
In the CPU Notifications, I get alerts even when CPU spike for a fraction of a second.
It would be great if I could say "alert me if >90% for 10 seconds or more."
Thanks!

[image]

wowhao333, 2025-07-04:
Hope this feature can be developed ASAP. 

It is a terrible experience to have multiple reminders in a few seconds.

@exelban Thanks!

eliseumds, 2025-12-30:
@nRanzo I sometimes run compilation/linting/formatting jobs that may take minutes to complete, so it'd be great if it was configurable, either per duration or process names (allowlist/denylist).

nRanzo, 2026-01-02:
@eliseumds allowing users to customize that threshold is reasonable, whereas tracking a single process feels borderline in terms of complexity vs. benefit. Users who frequently run heavy workloads will often have multiple processes that legitimately trigger the alert; in such cases, it makes more sense to disable the alert altogether.

The two most reasonable options, in my view, are making the threshold configurable, or letting users turn the alert off if they are not interested / already aware of the heavy workload running on their system.

Per-process tracking could work, but if users need to manage specific processes each time, disabling alerts during heavy workloads is simpler and more straightforward.