exelban/stats #2440: A leftover dot coming in the speed widget of disks and network.

Satya1543, 2025-03-12:
**Describe the bug**
In the disks and network section when I select transparent when no activity in the speed widget some part of it remains like a dot and when click on that dot it disapperas and after some time again comes.

Pls look into this issue it is coming from the very first version of "Stats". I am attaching a video for more clarity.

https://github.com/user-attachments/assets/f68ecce9-cbf0-43dd-9306-fb6f59d5344a

**Details:**
 - Device: [e.g. Macbook Air 2020]
 - macOS: [e.g. 14.7.4]
 - Application version: [e.g. 2.11.34]

Satya1543, 2025-03-26:
Pls solve this issue soon

exelban, 2025-03-27:
@Satya1543 do you have enabled combined modules?
what version do you use?

Satya1543, 2025-03-31:
> [@Satya1543](https://github.com/Satya1543) do you have enabled combined modules? what version do you use?
v2.11.36

Satya1543, 2025-03-31:
> [@Satya1543](https://github.com/Satya1543) do you have enabled combined modules? what version do you use?

combined modules means ?

exelban, 2025-05-15:
Hi. I cannot reproduce this behavior. It's or macOS bug, or some 3d party tool cause that.

closed 2025-05-15 by exelban