exelban/stats #2442: [REQUEST] Option in Clock calendar to start week on Monday

dzg, 2025-03-13:
I would like to be able to have the pop calendar start on Monday not Sunday:
[image]

Like [ItsyCal](https://www.mowglii.com/itsycal/):

[image]

And native Calendar:

[image]

Thanks!

exelban, 2025-03-13:
calendar automatically generates dates based on the system region. You can change that in your system settings.

exelban, 2025-03-13:
[image]

closed 2025-03-13 by exelban