exelban/stats #2448: Add a setting where I can choose to automatically change to external disk when the disk is connected

Andrewzhang00GH, 2025-03-16:
I'm copying a lot of stuff to my external disk, however, the data showed in menu bar does not automatically change to the external disk. I need to change it manually every time.
Maybe add a setting where I can choose to automatically change to external disk when the disk is connected?  Or automatically change to the active disk?
I cannot think of any better solutions now, hoping this will help

exelban, 2025-03-16:
Sorry. But I don't see it as a useful function for a lot of people. It will take a look of effort (and time) to add and support that.

closed 2025-03-16 by exelban

Andrewzhang00GH, 2025-03-16:
> Sorry. But I don't see it as a useful function for a lot of people. It will take a look of effort (and time) to add and support that.

Understood, thanks for replying