exelban/stats #2449: High CPU usage causes crash

edeetee, 2025-03-17:
Firstly, love this app. Great having quick access to so much system info when I'm doing intensive tasks.

**Describe the bug**

I play Magic the Gathering Arena on my mac. Pretty reliably, once I play a few games and close, stats has closed itself in the background.

MTGA seems to have unusually high CPU usage so my theory is that the high CPU usage is somehow crowding out stats and causing it to get automatically closed by the mac system.

I'm happy to provide logs or other details, I couldn't see a logging location in my cursory search of the repo.

**Details:**
 - Device: MacBook Pro 14" (M1 Max)
 - macOS: macOS Sequoia (15.3.1)
 - Application version: 2.11.35

exelban, 2025-03-17:
hi, thx.

Please share the crash report

edeetee, 2025-03-17:
I don't get the regular macos crash popup, it just closes without fanfare. Is there a location that I can find a crash report for it?

exelban, 2025-03-17:
It could be found in Applications -> Utilities -> Console. Find the Crash Reports tab. There you must see a row called Stats.

edeetee, 2025-03-17:
Thanks, didn't know Console had nice organisation like this in the sidebar haha
<details>
<summary>Crash report</summary>

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [753]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.34 (681)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-17 21:26:35.2704 +1300
OS Version:            macOS 15.3.1 (24D70)
Report Version:        12
Anonymous UUID:        76C45A70-4D4B-D7FC-0F3E-A8B1A8EE93F6

Sleep/Wake UUID:       02F6ADE1-DF2F-4536-A127-52D737BE9790

Time Awake Since Boot: 73000 seconds
Time Since Wake:       40112 seconds

System Integrity Protection: enabled

Crashed Thread:        10

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00007d64467b6e40
Exception Codes:       0x0000000000000001, 0x00007d64467b6e40

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [753]

VM Region Info: 0x7d64467b6e40 is not in any region.  Bytes after previous region: 32315964485185  
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      owned unmapped memory        unmapped-unmapped     [ 14.4M] rw-/rw- SM=PRV  owned physical footprint (unmapped)
--->  
      UNUSED SPACE AT END

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x192e0af54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x192e1d604 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x192e13af8 mach_msg_overwrite + 480
3   libsystem_kernel.dylib        	       0x192e0b29c mach_msg + 24
4   CoreFoundation                	       0x192f34a4c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x192f332ac __CFRunLoopRun + 1212
6   CoreFoundation                	       0x192f32734 CFRunLoopRunSpecific + 588
7   HIToolbox                     	       0x19e4a1530 RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x19e4a7348 ReceiveNextEventCommon + 676
9   HIToolbox                     	       0x19e4a7508 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x196aaa848 _DPSNextEvent + 660
11  AppKit                        	       0x197410c24 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688
12  AppKit                        	       0x196a9d874 -[NSApplication run] + 480
13  Stats                         	       0x102e4f714 0x102e4c000 + 14100
14  dyld                          	       0x192acc274 start + 2840

Thread 1:: WebCore: Scrolling
0   libsystem_kernel.dylib        	       0x192e0af54 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x192e1d604 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x192e13af8 mach_msg_overwrite + 480
3   libsystem_kernel.dylib        	       0x192e0b29c mach_msg + 24
4   CoreFoundation                	       0x192f34a4c __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x192f332ac __CFRunLoopRun + 1212
6   CoreFoundation                	       0x192f32734 CFRunLoopRunSpecific + 588
7   CoreFoundation                	       0x192fad9d0 CFRunLoopRun + 64
[... 753 lines cut ...]
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "644114de41e7236e6177f9bd",
      "factorPackIds" : {

      },
      "deploymentId" : 240000013
    },
    {
      "rolloutId" : "5f72dc58705eff005a46b3a9",
      "factorPackIds" : {

      },
      "deploymentId" : 240000015
    }
  ],
  "experiments" : [

  ]
}
}

```
</details>

exelban, 2025-03-17:
thx, it happen by itself or after you open the details view?

edeetee, 2025-03-17:
It happens by itself. When I exit MTGA, the status bar icons are just all
missing.
-

0224998841
@optiphonic <http://instagram.com/optiphonic>
edt.nz
<http://edt.nz>

exelban, 2025-03-17:
ok, thx. I will make a debug build for you tomorrow to check what is happening. Would you like to run it and check it out?

edeetee, 2025-03-17:
Yup! I am happy to use a debug build and try to catch more detailed logs.
-

0224998841
@optiphonic <http://instagram.com/optiphonic>
edt.nz
<http://edt.nz>

C41M50N, 2025-03-21:
Same thing happened to me today.

**System Info**:
- MacBook Pro 14" - M2 Max
- macOS Sonoma 14.7.2
- Stats Version: 2.11.35

<details><summary>Crash Report</summary><p>

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Stats [45893]
Path:                  /Applications/Stats.app/Contents/MacOS/Stats
Identifier:            eu.exelban.Stats
Version:               2.11.35 (682)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2025-03-20 23:23:55.5217 -0400
OS Version:            macOS 14.7.2 (23H311)
Report Version:        12
Anonymous UUID:        82EAF4C4-16E6-0479-4B63-42E078F3C8D4

Sleep/Wake UUID:       C37495F2-B302-4931-8294-F57F87FB69A6

Time Awake Since Boot: 3900000 seconds
Time Since Wake:       7494 seconds

System Integrity Protection: enabled

Crashed Thread:        7  Dispatch queue: eu.exelban.Stats.Repeater

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x000000010116ae6c

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [45893]

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18a760dd4 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18a7735c4 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x18a7699a4 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x18a761158 mach_msg + 24
4   CoreFoundation                	       0x18a881660 __CFRunLoopServiceMachPort + 160
5   CoreFoundation                	       0x18a87ff24 __CFRunLoopRun + 1208
6   CoreFoundation                	       0x18a87f414 CFRunLoopRunSpecific + 608
7   HIToolbox                     	       0x19502a19c RunCurrentEventLoopInMode + 292
8   HIToolbox                     	       0x195029fd8 ReceiveNextEventCommon + 648
9   HIToolbox                     	       0x195029d30 _BlockUntilNextEventMatchingListInModeWithFilter + 76
10  AppKit                        	       0x18e0decc8 _DPSNextEvent + 660
11  AppKit                        	       0x18e8d54d0 -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 700
12  AppKit                        	       0x18e0d1ffc -[NSApplication run] + 476
13  Stats                         	       0x100aaf6f4 0x100aac000 + 14068
14  dyld                          	       0x18a417154 start + 2476

Thread 1:: com.apple.coreanimation.render-server
0   libsystem_kernel.dylib        	       0x18a760dd4 mach_msg2_trap + 8
1   libsystem_kernel.dylib        	       0x18a7735c4 mach_msg2_internal + 80
2   libsystem_kernel.dylib        	       0x18a7699a4 mach_msg_overwrite + 476
3   libsystem_kernel.dylib        	       0x18a761158 mach_msg + 24
4   QuartzCore                    	       0x192abbb70 CA::Render::Server::server_thread(void*) + 580
5   QuartzCore                    	       0x192abb91c thread_fun(void*) + 32
6   libsystem_pthread.dylib       	       0x18a7a1f94 _pthread_start + 136
7   libsystem_pthread.dylib       	       0x18a79cd34 thread_start + 8

Thread 2:: WebCore: Scrolling
[... 660 lines cut ...]
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "661464ecda55e5192b100804",
      "factorPackIds" : {

      },
      "deploymentId" : 240000005
    },
    {
      "rolloutId" : "64628732bf2f5257dedc8988",
      "factorPackIds" : {

      },
      "deploymentId" : 240000001
    }
  ],
  "experiments" : [

  ]
}
}

```
</p></details>

exelban, 2025-03-21:
hi, please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/19388064/Stats.dmg.zip)

C41M50N, 2025-03-24:
seems to work. its been stable for over 48 hours now. thanks!

exelban, 2025-03-24:
nice, the latest release contains that fix. So you can use latest release. Ping me if it will repeat.

closed 2025-03-24 by exelban