exelban/stats #2456: Add sensor to display CPU speed limit

jonatan-ivanov, 2025-03-18:
`CPU_Speed_Limit` can tell if the CPU is (thermal) throttled which can be useful information to display especially since Intel Macs have a tendency to degrade with thermal throttling over time.

`pmset -g thermlog` can tell this information and on my 2019 Mac, I can definitely see this is going under 100% (the CPU is throttled to prevent overheating) under load.

This is the same as https://github.com/exelban/stats/issues/193 which was close due to this information is not useful. I think it is especially if the Mac is not new.
Right now one can use `pmset -g thermlog` or [Hot](https://github.com/macmade/Hot) which has two main features: showing the CPU temperature and the speed limit. Having 2.6k stars might prove that people find this information useful.

exelban, 2025-03-18:
There is this information. But since it's available only on intel I cannot even test if it still works. But the code is there and it was working.

jonatan-ivanov, 2025-03-18:
Could you please check if `sudo powermetrics -s thermal` (for Apple Silicon) works for you?
For testing (not sure if you want to go that way) GH Actions offers Intel-based Mac runners.

I'm not sure what you mean by that the code is there and it was working. What should I do with this information as a user? I don't see this info available on the UI. Did you mean I can fork and build it on my own? (In that case, for me it's easier to just use [Hot](https://github.com/macmade/Hot) instead.)