exelban/stats #2457: Bluetooth battery is wrong

webfrank, 2025-03-19:
I've enabled bluetooth battery monitoring for keyboard and magic mouse but values reported by widget are not the same reported by system.

[image]

[image]

exelban, 2025-03-22:
Can you share the output of `ioreg -r -c AppleDeviceManagementHIDEventService | grep -E "BatteryPercent|Product"`?

webfrank, 2025-03-23:
[image]

[image]

It seems the percentages are fixed

closed 2025-03-23 by exelban