exelban/stats #2468: Stats remote login screen keeps popping up when starting app

tjabbie, 2025-03-25:
Every time when I reboot my laptop, or just start the app manualy I see the Stats remote login screen. I have to press cancel. Didn't activate it in the settings. Because when I go in the settings and toggle the remote option I get the same screen.

Macbook air running 15.3.2 (24D2082)

closed 2025-03-25 by exelban

jimmylzt188, 2025-04-02:
export the setting ，edit the “remote_state” to “no” in “Stats.plist”，import the setting

exelban, 2025-04-02:
unfortunately there is no good way to disable this popup since it was enabled till this feature will be publicly available.
The best way is proposed by @jimmylzt188

exelban, 2025-04-02:
I have prepared a spacial build with changed logic a little bit. It should prevent from opening a login screen if not enabled:

[Stats.dmg.zip](https://github.com/user-attachments/files/19571074/Stats.dmg.zip)