exelban/stats #2470: Stats shows incorrect temperature (80.0°C) on MacBook Pro M1

eje-1, 2025-03-26:
Description:
I'm experiencing an issue with the Stats app on my MacBook Pro (M1, 2020). The CPU temperature displayed in the app is incorrect and does not change over time.

    The app shows a constant 80°C, even under different workloads.

    Running monitoring tools in the terminal shows 0.0°C (which might be another issue, but still suggests a reading problem).

    The temperature reading seems frozen and does not reflect actual CPU activity.

Steps to reproduce:

    Install and launch the Stats app.

    Enable CPU temperature monitoring.

    Observe that the value remains static (e.g., 80°C).

    Compare it with other system monitoring tools.

Expected behavior:
The CPU temperature should update dynamically and reflect real-time changes in system load.

Actual behavior:
The CPU temperature remains stuck at 80°C and does not change.

System Information:

    MacBook Pro (M1, 2020)

    macOS version: (15.3.2)

    Stats app version: (Version 2.11.36)

Additional Notes:

    This issue might be related to Apple Silicon Macs and how the app reads sensor data.

    Other monitoring tools also seem to struggle with correct temperature readings on M1 chips.

Would be great if the developer could look into this! Thanks.

[image]

exelban, 2025-03-26:
if you have M1 basically you must have no temperature here since it's not available on M1. For better understanding take a look at temperatures in the sensors module

eje-1, 2025-03-26:
All right, thank you for the answer, but I also have a Macbook Pro M1 Max 16" and I see the correct temp in the stats app. That's why I was wondering.

exelban, 2025-03-26:
M1 Max works fine since it has normal SMC sensors. And m1 not. But still, what sensors module shows for m1? Cause it strange that you have any value visible in the CPU module

exelban, 2025-03-26:
PS: you can have sensors by enabling option `HID sensors` in the sensors module