exelban/stats #2479: Settings dialog runs, but nothing in menu bar

briank-cxc, 2025-04-01:
I'm not sure if this is a bug or just me being feeble, but when I launch from the application folder, I get nothing. When I launch a second time, I get the settings dialog. Toggling any of the components on/off does nothing. I never see anything in the menu bar or elsewhere on my screen.

[image]

I'm not sure where to find logs, but if I launch the executable directly from Terminal, I see this running log:

```
% /Applications/Stats.app/Contents/MacOS/Stats 
2025-04-01 08:15:46 reader.swift:90 DBG [<CPU.LoadReader: 0x12a60a680>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:166 DBG [<CPU.ProcessReader: 0x6000000c50e0>] Set update interval: 1 sec
2025-04-01 08:15:46 reader.swift:90 DBG [<CPU.ProcessReader: 0x6000000c50e0>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:166 DBG [<CPU.AverageReader: 0x6000000c5440>] Set update interval: 60 sec
2025-04-01 08:15:46 reader.swift:90 DBG [<CPU.AverageReader: 0x6000000c5440>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:90 DBG [<CPU.TemperatureReader: 0x6000000c54d0>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:90 DBG [<CPU.FrequencyReader: 0x6000007cd380>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:90 DBG [<GPU.InfoReader: 0x6000003cada0>] Successfully initialize reader
2025-04-01 08:15:46.650 Stats[5793:684849] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2025-04-01 08:15:46 reader.swift:90 DBG [<RAM.UsageReader: 0x600000ec8800>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:166 DBG [<RAM.ProcessReader: 0x6000000f8240>] Set update interval: 1 sec
2025-04-01 08:15:46 reader.swift:90 DBG [<RAM.ProcessReader: 0x6000000f8240>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:90 DBG [<Disk.CapacityReader: 0x6000000dc510>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:166 DBG [<Disk.ActivityReader: 0x6000000f9830>] Set update interval: 1 sec
2025-04-01 08:15:46 reader.swift:90 DBG [<Disk.ActivityReader: 0x6000000f9830>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:166 DBG [<Disk.ProcessReader: 0x6000000f86c0>] Set update interval: 1 sec
2025-04-01 08:15:46 reader.swift:90 DBG [<Disk.ProcessReader: 0x6000000f86c0>] Successfully initialize reader
2025-04-01 08:15:46 reader.swift:90 DBG [<Sensors.SensorsReader: 0x600000bf6920>] Successfully initialize reader
2025-04-01 08:15:47 readers.swift:317 DBG [<Sensors.SensorsReader: 0x600000bf6920>] Found 2 fans
2025-04-01 08:15:47.694 Stats[5793:684849] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2025-04-01 08:15:47 reader.swift:90 DBG [<Net.UsageReader: 0x11d8cf2a0>] Successfully initialize reader
2025-04-01 08:15:47 reader.swift:90 DBG [<Net.ProcessReader: 0x6000003f0fa0>] Successfully initialize reader
2025-04-01 08:15:47 reader.swift:90 DBG [<Net.ConnectivityReader: 0x12a6ad070>] Successfully initialize reader
2025-04-01 08:15:47 reader.swift:90 DBG [<Battery.UsageReader: 0x11d8fc710>] Successfully initialize reader
2025-04-01 08:15:47 reader.swift:90 DBG [<Battery.ProcessReader: 0x600001d20000>] Successfully initialize reader
2025-04-01 08:15:47 reader.swift:90 DBG [<Bluetooth.DevicesReader: 0x6000005c9e40>] Successfully initialize reader
2025-04-01 08:15:47 reader.swift:90 DBG [<Clock.ClockReader: 0x600000084480>] Successfully initialize reader
Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process." UserInfo={NSDebugDescription=The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process.}
2025-04-01 08:15:47 widget.swift:296 DBG [Battery] widget battery enabled
2025-04-01 08:15:47 reader.swift:135 DBG [<Net.UsageReader: 0x11d8cf2a0>] Set up update interval: 1 sec
2025-04-01 08:15:47 reader.swift:135 DBG [<Net.ConnectivityReader: 0x12a6ad070>] Set up update interval: 1 sec
2025-04-01 08:15:47 widget.swift:296 DBG [Network] widget speed enabled
2025-04-01 08:15:47 reader.swift:135 DBG [<Disk.CapacityReader: 0x6000000dc510>] Set up update interval: 1 sec
2025-04-01 08:15:47 reader.swift:135 DBG [<Disk.ActivityReader: 0x6000000f9830>] Set up update interval: 1 sec
2025-04-01 08:15:47 widget.swift:296 DBG [Disk] widget mini enabled
2025-04-01 08:15:47 reader.swift:135 DBG [<RAM.UsageReader: 0x600000ec8800>] Set up update interval: 1 sec
2025-04-01 08:15:47 widget.swift:296 DBG [RAM] widget mini enabled
2025-04-01 08:15:47 reader.swift:135 DBG [<CPU.LoadReader: 0x12a60a680>] Set up update interval: 1 sec
2025-04-01 08:15:47 reader.swift:135 DBG [<CPU.FrequencyReader: 0x6000007cd380>] Set up update interval: 1 sec
2025-04-01 08:15:47 widget.swift:296 DBG [CPU] widget line_chart enabled
2025-04-01 08:15:47 helpers.swift:220 DBG The support window was shown 2 days ago, stopping...
2025-04-01 08:15:47 helpers.swift:112 DBG Application update interval is 'Silent'
2025-04-01 08:15:47 AppDelegate.swift:84 INF Stats started in 0.0137 seconds
```

I'm using:
Stats 2.11.37
MacOS 15.4
Macbook Pro M2

I use multiple Spaces if that factors into this issue.

exelban, 2025-04-01:
Most probably you have some tools that hiding items in the menu bar. According to the logs there are modules and widgets enabled.

closed 2025-04-06 by exelban

briank-cxc, 2025-04-08:
Yes, I had an old version of Menu Meters that I didn't realize was starting at login. After closing and restarting, everything is working as it should.