exelban/stats #2484: Fleeting notifications

rebop, 2025-04-02:
Hi there,

I have Stats notifications set to ALERTS so they should stay on the screen. But I am getting notifications last few days that pop up and vanish so quick cannot be read. I have no idea what they aare alerting me to.

Can we fix? Is there a log I can review to see what they were?

Thanks,~Bob

exelban, 2025-04-03:
Hi. Could you please specify what notification you have enabled?
Cause for example if it's a RAM pressure level for example that just jumps to critical state for a second, then notification will be shown -> and after a sec when RAM goes to the normal state it will be removed.

rebop, 2025-04-03:
Thanks for jumping in. This has happened again twice since I posted.

In System Settings, all are enabled with Alerts selected, rather than Banner. So the message should be persistent. But it closes before I can even read it!! It appears it might say CPU or GPU but either would be a surprise as nothing strenuous happening when this occurs. Sometime on closing an app. Sometimes nothing unusual.

In STATS:

CPU 75%
GPU 75%
RAM 75%
Fans 50%

That's all. If it is recommended to enable more, please advise.

~Bob

rebop, 2025-04-03:
Well, that is true!!!

Just happened. All I caught was GPU for sure. I can tell you I have never seen this before the last Stats update. Something has changed, at least for me. Maybe I am using more GPU, but I cannot imagine why. This time I only opened an app to get the notification.

Thoughts???

exelban, 2025-04-03:
nothing was changed to the notifications for months, so maybe some changes in the macOS

rebop, 2025-04-03:
Nothing in MACOS. And I did get a very recent Stats update. Seem very coincidental that this started about then. I did not take exact note.

This is 2.11.37. When was that released?

rebop, 2025-04-03:
Definitely GPU warning Just got 4 in a row!!! After clsing an app. I do not belive it is true that I jumped to 76% GPU FOUR times after closing an app. Would you not agree?

rebop, 2025-04-03:
And if the notification closes before I can even read it, not very useful, is it? Why not persistent like most (all?) other notifications set to alert?

closed 2025-04-06 by exelban