exelban/stats #2487: won't run on MacOS 15.4

jauling, 2025-04-04:
**Describe the bug**
Upgraded to MacOS 15.4 Sequoia yesterday (from 15.3.2). I have Stats.app in my Login Items, but I noticed it doesn't run on boot anymore. When I doubleclick on Stats.app, the splash window displays for a split second then disappears.

**Details:**
 - Device: [Macbook Air M1]
 - macOS: [15.4]
 - Application version: [2.11.37] via Homebrew

jauling, 2025-04-04:
Running from Terminal:

```
% /Applications/Stats.app/Contents/MacOS/Stats 
2025-04-04 11:13:31 reader.swift:85 DBG [<CPU.LoadReader: 0x15c830ed0>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:159 DBG [<CPU.ProcessReader: 0x600002d07a80>] Set update interval: 3 sec
2025-04-04 11:13:31 reader.swift:85 DBG [<CPU.ProcessReader: 0x600002d07a80>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:159 DBG [<CPU.AverageReader: 0x600002d18ab0>] Set update interval: 60 sec
2025-04-04 11:13:31 reader.swift:85 DBG [<CPU.AverageReader: 0x600002d18ab0>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:85 DBG [<CPU.TemperatureReader: 0x600002d18b40>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:85 DBG [<CPU.FrequencyReader: 0x600002809ce0>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:85 DBG [<GPU.InfoReader: 0x600002d0a130>] Successfully initialize reader
2025-04-04 11:13:31.773 Stats[61490:2035003] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2025-04-04 11:13:31 reader.swift:85 DBG [<RAM.UsageReader: 0x60000232c400>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:159 DBG [<RAM.ProcessReader: 0x600002d29cb0>] Set update interval: 1 sec
2025-04-04 11:13:31 reader.swift:85 DBG [<RAM.ProcessReader: 0x600002d29cb0>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:85 DBG [<Disk.CapacityReader: 0x60000316bf00>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:159 DBG [<Disk.ActivityReader: 0x60000318c180>] Set update interval: 1 sec
2025-04-04 11:13:31 reader.swift:85 DBG [<Disk.ActivityReader: 0x60000318c180>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:159 DBG [<Disk.ProcessReader: 0x600002d25290>] Set update interval: 1 sec
2025-04-04 11:13:31 reader.swift:85 DBG [<Disk.ProcessReader: 0x600002d25290>] Successfully initialize reader
2025-04-04 11:13:31 reader.swift:85 DBG [<Sensors.SensorsReader: 0x600002e173e0>] Successfully initialize reader
2025-04-04 11:13:32.248 Stats[61490:2035003] NSPanel requires NSWindowStyleMaskUtilityWindow for a HUD window
2025-04-04 11:13:32 reader.swift:85 DBG [<Net.UsageReader: 0x12c94f4d0>] Successfully initialize reader
2025-04-04 11:13:32 reader.swift:85 DBG [<Net.ProcessReader: 0x600002d48630>] Successfully initialize reader
2025-04-04 11:13:32 reader.swift:85 DBG [<Net.ConnectivityReader: 0x15aebd000>] Successfully initialize reader
2025-04-04 11:13:32 reader.swift:85 DBG [<Battery.UsageReader: 0x12ca56f60>] Successfully initialize reader
2025-04-04 11:13:32 reader.swift:85 DBG [<Battery.ProcessReader: 0x6000031ede00>] Successfully initialize reader
2025-04-04 11:13:32 reader.swift:85 DBG [<Bluetooth.DevicesReader: 0x600002e24c80>] Successfully initialize reader
2025-04-04 11:13:32 reader.swift:85 DBG [<Clock.ClockReader: 0x6000031eba00>] Successfully initialize reader
Error Domain=NSCocoaErrorDomain Code=4099 "The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process." UserInfo={NSDebugDescription=The connection to service named eu.exelban.Stats.SMC.Helper was invalidated: failed at lookup with error 3 - No such process.}
2025-04-04 11:13:32 reader.swift:128 DBG [<Net.UsageReader: 0x12c94f4d0>] Set up update interval: 1 sec
2025-04-04 11:13:32 reader.swift:128 DBG [<Net.ConnectivityReader: 0x15aebd000>] Set up update interval: 1 sec
2025-04-04 11:13:32 widget.swift:296 DBG [Network] widget speed enabled
2025-04-04 11:13:32 reader.swift:128 DBG [<Sensors.SensorsReader: 0x600002e173e0>] Set up update interval: 1 sec
2025-04-04 11:13:32 widget.swift:296 DBG [Sensors] widget mini enabled
2025-04-04 11:13:32 reader.swift:128 DBG [<CPU.LoadReader: 0x15c830ed0>] Set up update interval: 3 sec
2025-04-04 11:13:32 reader.swift:128 DBG [<CPU.FrequencyReader: 0x600002809ce0>] Set up update interval: 3 sec
2025-04-04 11:13:32 widget.swift:296 DBG [CPU] widget mini enabled
2025-04-04 11:13:32 helpers.swift:110 DBG Application update interval is 'At start'
2025-04-04 11:13:32 helpers.swift:151 DBG error updater.check(): last check was 8 minutes ago, stopping...
2025-04-04 11:13:32 AppDelegate.swift:85 INF Stats started in 0.0257 seconds
zsh: trace trap  /Applications/Stats.app/Contents/MacOS/Stats
```

exelban, 2025-04-04:
please share the crash report

jauling, 2025-04-04:
[Stats-2025-04-04-111334.ips.zip](https://github.com/user-attachments/files/19602334/Stats-2025-04-04-111334.ips.zip)

exelban, 2025-04-04:
please update to the latest release

closed 2025-04-04 by exelban