exelban/stats #2492: Stats crashes after a few hours

uhkeller, 2025-04-07:
**Describe the bug**
After a few hours of working fine, Stats will suddenly close without any error message. There are Crash Reports in the Console. I'll attach the latest one.

**Details:**
 - Device: M1 Macbook Pro (2021)
 - macOS: 15.4
 - Application version: 2.11.39

[Stats-crash.txt](https://github.com/user-attachments/files/19635655/Stats-crash.txt)

exelban, 2025-04-08:
Please check this build and please give a feedback if it works or not:

[Stats.dmg.zip](https://github.com/user-attachments/files/19651374/Stats.dmg.zip)

uhkeller, 2025-04-08:
Thanks. I'm running it now and will report back.

uhkeller, 2025-04-10:
Still crashes unfortunately. I didn't respond immediately because it can take a day or so for it to crash again. Here's the log.

[Stats-crash2.txt](https://github.com/user-attachments/files/19678872/Stats-crash2.txt)

exelban, 2025-04-10:
Yeap, but it's totally different crash reason, will take a look what could be done here.

exelban, 2025-04-11:
hi, please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/19706859/Stats.dmg.zip)

uhkeller, 2025-04-11:
Thanks! Just installed it.

exelban, 2025-04-12:
@uhkeller hi, what about the last build?

uhkeller, 2025-04-12:
So far so good, but I haven't used the machine much since I installed it. I'll be mostly AFK until Monday.

uhkeller, 2025-04-14:
I've had it running for the whole day, no crash so far!

exelban, 2025-04-14:
you can install the latest release. It contains all fixes that was in the build I share here.

closed 2025-04-15 by exelban