exelban/stats #2512: Software version 2.11.39 will flash back

Ztt-87, 2025-04-13:
Using this version on MacBook AIR will inexplicably flash back, and you need to restart the program, but after a period of time, the software will flash back again.

macOS 15.3.2 (24D81)

Chip: Apple M3

exelban, 2025-04-13:
what do you mean by "flash back"?

Ztt-87, 2025-04-15:
During the running of the Stats application, it will automatically exit. You need to reopen the application, but it will automatically exit after a while.

Ztt-87, 2025-04-15:
> 你说的“闪回”是什么意思？

During the running of the Stats application, it will automatically exit. You need to reopen the application, but it will automatically exit after a while.

exelban, 2025-04-15:
First of all: what version of Stats do you use?
Please share the crash report

Ztt-87, 2025-04-15:
> 请分享崩溃报告

OK, please wait a moment. I will send it to you after I get the crash report.

Ztt-87, 2025-04-16:
> Please share the crash report

crash report
[Stats_log.zip](https://github.com/user-attachments/files/19769025/Stats_log.zip)

exelban, 2025-04-16:
please update to the latest release

Ztt-87, 2025-04-16:
> please update to the latest release

OK, I'll try to update the new version and give you feedback.