exelban/stats #2514: MacBook Air (M3) macOS Sequoia (15.4.0) no module shows up in the top bar

bitcoinvsalts, 2025-04-14:
is it normal?

`<?xml version="1.0" encoding="UTF-8"?>
<!DOCTYPE plist PUBLIC "-//Apple//DTD PLIST 1.0//EN" "http://www.apple.com/DTDs/PropertyList-1.0.dtd">
<plist version="1.0">
<dict>
	<key>Battery_position</key>
	<integer>7</integer>
	<key>Battery_state</key>
	<true/>
	<key>Bluetooth_position</key>
	<integer>5</integer>
	<key>CPU_barChart_position</key>
	<integer>3</integer>
	<key>CPU_label_position</key>
	<integer>1</integer>
	<key>CPU_lineChart_position</key>
	<integer>2</integer>
	<key>CPU_line_chart_color</key>
	<string>white</string>
	<key>CPU_line_chart_label</key>
	<true/>
	<key>CPU_line_chart_value</key>
	<true/>
	<key>CPU_line_chart_valueColor</key>
	<true/>
	<key>CPU_mini_alignment</key>
	<string>right</string>
	<key>CPU_mini_color</key>
	<string>yellow</string>
	<key>CPU_mini_position</key>
	<integer>0</integer>
	<key>CPU_oneView</key>
	<true/>
	<key>CPU_pieChart_position</key>
	<integer>4</integer>
	<key>CPU_position</key>
	<integer>0</integer>
	<key>CPU_state</key>
	<true/>
	<key>CPU_tachometer_position</key>
	<integer>5</integer>
	<key>CPU_widget</key>
	<string>label,mini,line_chart</string>
	<key>Clock_position</key>
	<integer>8</integer>
	<key>CombinedModules</key>
	<true/>
	<key>CombinedModules_spacing</key>
	<string>1</string>
	<key>Disk_position</key>
	<integer>3</integer>
	<key>Disk_state</key>
	<true/>
	<key>GPU_barChart_position</key>
	<integer>3</integer>
	<key>GPU_label_position</key>
	<integer>2</integer>
	<key>GPU_lineChart_position</key>
	<integer>0</integer>
	<key>GPU_mini_position</key>
	<integer>1</integer>
	<key>GPU_position</key>
	<integer>1</integer>
	<key>GPU_state</key>
	<true/>
	<key>GPU_tachometer_position</key>
	<integer>4</integer>
	<key>GPU_widget</key>
	<string>mini,line_chart</string>
	<key>LaunchAtLoginNext</key>
	<true/>
	<key>NSToolbar Configuration eu.exelban.Stats.Settings.Toolbar</key>
	<dict>
		<key>TB Display Mode</key>
		<integer>1</integer>
		<key>TB Icon Size Mode</key>
		<integer>1</integer>
		<key>TB Is Shown</key>
		<true/>
		<key>TB Size Mode</key>
		<integer>1</integer>
	</dict>
	<key>NSWindow Frame NSNavPanelAutosaveName</key>
	<string>640 696 430 163 0 0 1710 1068 </string>
	<key>Network_position</key>
	<integer>6</integer>
	<key>Network_state</key>
	<true/>
	<key>RAM_position</key>
	<integer>2</integer>
	<key>RAM_state</key>
	<true/>
	<key>Sensors_barChart_position</key>
	<integer>3</integer>
	<key>Sensors_label_position</key>
	<integer>2</integer>
	<key>Sensors_mini_position</key>
	<integer>1</integer>
	<key>Sensors_oneView</key>
	<true/>
	<key>Sensors_position</key>
	<integer>4</integer>
	<key>Sensors_sensors_alignment</key>
	<string>right</string>
	<key>Sensors_stack_position</key>
	<integer>0</integer>
	<key>Sensors_state</key>
	<true/>
	<key>Sensors_unknown</key>
	<true/>
	<key>Sensors_widget</key>
	<string>mini,sensors</string>
	<key>dockIcon</key>
	<true/>
	<key>pause</key>
	<false/>
	<key>runAtLoginInitialized</key>
	<true/>
	<key>sensor_Average CPU</key>
	<true/>
	<key>setupProcess</key>
	<true/>
	<key>support_ts</key>
	<integer>1744644321</integer>
	<key>telemetry</key>
	<false/>
	<key>temperature_units</key>
	<string>celsius</string>
	<key>updater_check_ts</key>
	<integer>1744644238</integer>
	<key>version</key>
	<string>2.11.40</string>
</dict>
</plist>
`

closed 2025-04-14 by exelban

bitcoinvsalts, 2025-04-15:
I cannot find the duplicate. Am I missing something?

exelban, 2025-04-15:
for 99% you have some sort of app that manipulates the menu bar and it hides the widgets or you have enabled 0 modules.