exelban/stats #2515: [Feature] Disk Free Space in GB/TB

romanr, 2025-04-15:
To have plain free space without total and not in percent free space.  
There's Free space widget but it displays total and free space is too small to read at a glance. as half as small as it could be.   
I don't know who needs to know total, it's always same number.  

(And there's Free space in percent. To know how much is 1% or 3% - you have to do mental mathematics to make use of this number.)

exelban, 2025-04-15:
what do you mean `displays total and free space` ?
how can you visualize the total space of the disk in the menu bar?

romanr, 2025-04-15:
I didn't know all the terminology of the project, sorry.  I mean Mini widget. Here is the prototype:

[image]

romanr, 2025-04-15:
I can implement it myself.

By the way, The "Colour - based on utilisation"

[image]

Does not apply to modern disks when utilisation measured in percent. When disks were small, yes. But now with terabyte sized disks, 5% of 1TB  may not necessarily be critical displayed in red. it's 50GB. While in the past, 5% of 10GB disk free is alarming.
Configurable threshold like in iStat would work here.

closed 2025-04-28 by exelban