exelban/stats #2518: Abnormal exit of the software

Ztt-87, 2025-04-20:
你好，打扰一下！上周我反馈了一个软件异常退出的问题

exelban, 2025-04-20:
hi, please disable the network module and check if Stats will crash again

Ztt-87, 2025-04-21:
> hi, please disable the network module and check if Stats will crash again

OK, I'll disable the network module and observe it.

Ztt-87, 2025-04-23:
> hi, please disable the network module and check if Stats will crash again

The network module was disabled and observed for two days, and there was no crash.

Ztt-87, 2025-04-23:
> ok, thx. What widget have you enabled in the network module?

[image]
[image]
[image]

exelban, 2025-04-23:
hmm, could you disable this one you have enabled and enable for example NET widget?
After enable the network module, wonder if it's a widget causes the crash.

Ztt-87, 2025-04-23:
> hmm, could you disable this one you have enabled and enable for example NET widget? After enable the network module, wonder if it's a widget causes the crash.

Excuse me, is it closed like this?
I have closed the widget now. Do I need to observe it for a while?
[image]

exelban, 2025-04-23:
yeap, please observe that for some time. You can enable the label widget (the one that just shows NET). Cause it for sure cannot cause the crash

Ztt-87, 2025-04-24:
> yeap, please observe that for some time. You can enable the label widget (the one that just shows NET). Cause it for sure cannot cause the crash

OK, I have enabled the label widget (the one that only shows NET). I will observe for a while and give you feedback.

[image]

Ztt-87, 2025-04-25:
> yeap, please observe that for some time. You can enable the label widget (the one that just shows NET). Cause it for sure cannot cause the crash

I enabled the label widget (only the widget that shows NET). After observing for a period of time, there was no collapse. 

Could you help me find out why turning on network widgets will cause the program to crash?🤝

Ztt-87, 2025-05-27:
> ok, thx. Yeap, I will make a debug build that helps to find what exactly it crashes

Hello, is there a solution to this problem at present? What other logs do I need to provide you?

closed 2025-05-27 by exelban

Ztt-87, 2025-05-27:
> [@Ztt-87](https://github.com/Ztt-87) hi, sorry for not response. I'm pretty sure it's the same problem as in [#2550](https://github.com/exelban/stats/issues/2550) please take a look at the latest build I have shared there

OK, I'll try to install it. I see that the software version is 2.11.43. I tried to install this version yesterday, but there was still an abnormal exit. I reinstalled it once. I'll observe it.