exelban/stats #2523: [Bug]Chinese characters are not displayed correctly

anyerqi, 2025-04-24:
**Describe the bug**
When viewing the `Top processes`, Chinese characters are not displayed correctly — they appear as garbled symbols (e.g., `M-dM-=M^S`). This issue makes it difficult to read process names or command-line arguments in Chinese. It seems to be an encoding problem.

**Details:**
 - Device: MacBook Air M1, 2020
 - macOS: 15.3.2
 - Application version: 2.11.41

[image]

exelban, 2025-04-24:
check what `/bin/ps -Aceo pid,pcpu,comm -r` returns

anyerqi, 2025-04-25:
> /bin/ps -Aceo pid,pcpu,comm -r

in my terminal, the output is OK:

[image]

in my opinion, there is the env:`LANG=en_US.UTF-8`, so it's OK in my terminal.

closed 2025-06-22 by exelban