exelban/stats #2531: wrong free disk space (Stats 2.11.42)

checktext00, 2025-04-30:
**Describe the bug**
Since Stats 2.11.42 (feat: added a new mechanism for disk free space reader that potentially will take less CPU to make a read), the free disk space reading is wrong on APFS and FAT32 formatted volumes

for the APFS volume it's showing as 90% free space even though it should be 10%
for the FAT32 it's showing 200% free space (twice what the volume can hold)

**Details:**
 - macOS: 11
 - Application version: 2.11.42

exelban, 2025-04-30:
share the screenshot of available space in stats and in disk manager

checktext00, 2025-04-30:
can't reproduce it with the exFAT volume, but for the APFS volume that macOS is booted from Disk Utility shows
APFS Container
Capacity: 251 GB
Free: 24 GB
Used: 227 GB

and Stats 2.11.42 shows
237 GB of 251 GB free, or after clicking it 14 GB of 251 GB used

Stats 2.11.41 shows
25 GB of 251 GB free, or 226 GB of 251 GB used

booxter, 2025-05-04:
I may hit a similar issue, though not sure since I'm on 2.11.41. I'm a new user.

[image]

[image]

[image]

Note there's a separate volume (for Nix store), maybe it gets in the way somehow.

booxter, 2025-05-04:
@exelban for me, yes. No change that I can see (both show what I would consider wrong info).

exelban, 2025-05-04:
so it's not related to the changes in the latest version. For 99% it's a macOS reads, there are a few problems with getting free disk space accurate. Since macOS uses a lot of tricks with purgable space(

closed 2025-05-05 by exelban

booxter, 2025-05-05:
@checktext00 I don't think it matches what the widget shows anyway, see:

[image]

[image]

booxter, 2025-05-05:
@exelban I see you closed the issue as Not planned. I respect it, it's your project. But so that I understand: do you think the current behavior of the disk widget is correct? Or is the project not getting updates anymore? Any other reason?

exelban, 2025-05-05:
@booxter if you have the same value in v2.11.41 as v2.11.42 I cannot help with that. I literally do not know another way to obtain disk space.

Also, as you notice you have the same value in *41 as in *42, so your for 100% does not touched by this issue.

booxter, 2025-05-05:
Thanks. I will report another issue then.