exelban/stats #2533: [Question/Possible issue] How to include config in dotfiles? Import/export?

jadnhm, 2025-04-30:
[image]

I don't seem to be able to get the 'Stats Remote' thing to work at all, whatever it's supposed to do... maybe I need to be included in some kind of beta list or something first?... The button takes me to a site but when I try to 'sign up' I get this 'invalid verification' message

[image]

[image]

[image]

[image]

-----

I've tried to use the Export/Import functionality, but it always ends up totally resetting all my settings. 

----

All of this is in service of 'maintaining my settings' through a catastrophic computer event (system crash, new computer, setting up multiple computers, etc) and what I really want is to store the settings file in my dotfiles repo somehow. 

I'm currently using [YADM](https://yadm.io) and I tried to set this up a couple of different ways and none of them seemed to work:
a] copy the 'in use' settings file (`~/Library/Preferences/eu.exelban.Stats.plist` <-- I think that's right?) into my dotfiles (`~/.config/exelban_stats/Stats.plist`) and linking it back into place
b] do the 'export' function and save the file to my dotfiles (see above) and try to link it in place
    Note: this one seems to have exported an XML plist and the program wants a binary plist? 
c] I also tried converting the exported XML plist to binary using something like `plutil -convert binary1 -o ~/.config/exelban_stats/Stats-binary.plist ~/.config/exelban_stats/Stats-xml.plist` and then linking THAT into place (again see above)

These all result in Stats starting with 'no settings' and taking me through the little setup wiz thing. 

Any help for me would be great! Docs that explain what's up here would be better. If I can get it working I would be happy to submit a PR with some docs about this.

exelban, 2025-04-30:
For now I just have not enough time to complete remote fully working. Maybe will try to finish that this week.

What do you mean that export/import reset the settings?
I recheck everything right now and it works properly. Do you have any message? Did the saving window appear after you click save?

exelban, 2025-05-01:
`~/Library/Preferences/eu.exelban.Stats.plist` - it's a Stats preferences basically. Export/import just saves/loads that file. So you can work with that as you want. I really don't know how macOS will handle links, but you can try.

Im pretty sure you have some old macOS, and it's the reason why import/export does not work

closed 2025-05-07 by exelban