exelban/stats #2534: Menu icon disappears when moving or switching to Combined mode

cepheid666, 2025-05-01:
**Describe the bug**
On a fresh install of Stats v2.11.42, the only menu-bar icon to appear is Battery even though other modules are active.  Switching to "Combined modules" makes the menu-bar icon disappear entirely and it will not return until the settings are reset.  Moving the icon to the left (on the other side of the notch) also makes it disappear.

Steps to reproduce:
1) Open a fresh install of Stats
2) Notice the menu-bar icon (see screenshot 1 -- the battery icon is the only one that shows up, on the far left, which is just to the right of the notch on my screen)
[image]
3) In the Stats settings, enable "Combined modules" ...
4) The menu-bar icon has now disappeared and is no longer retrievable even when disabling "Combined modules" (see screenshot 2).
[image]
5) The only way to get this back is to reset all the settings.

**Details:**
 - Device: MacBook Pro 16-inch, 2023, M2 Max
 - macOS: Ventura 13.7.5
 - Application version: 2.11.42

Help is appreciated!!

cepheid666, 2025-05-01:
Could I ask what issue this is a duplicate of? I searched before reporting to try not to waste your time but clearly I missed it… would like to follow that issue.

Thanks!

exelban, 2025-05-01:
looking to the number of items in your menu bar I'm for 100% sure that combined widget just cannot fit in your menu bar

cepheid666, 2025-05-02:
Can’t items fit to the left of the notch? I thought the menu items could extend past the notch, but maybe I am wrong in that?

Thanks!

exelban, 2025-05-02:
nope, it cannot