exelban/stats #2539: Remote monitor - stuck on even with switch off

scottnzuk, 2025-05-04:
So every power up i get the remote box, asking for login, i think this started because i swiped the switch but then didn't setup and it appears to be off in menu, but still asks every login.

closed 2025-05-04 by exelban

scottnzuk, 2025-05-04:
?

scottnzuk, 2025-05-04:
Hi @exelban 
Just wondering if there was a specific reason the bug ticket was closed without any comment? If it was missing required details or not in the correct format, I’m happy to update it accordingly. I really like the product and appreciate your work, but I have to admit that some of the past responses have felt a bit dismissive. I had actually been considering donating, but this kind of interaction makes me rethink continuing with the tool.

Thanks for your time.

exelban, 2025-05-04:
please first of all, read the FAQ. Ask again if there will be still questions

scottnzuk, 2025-05-04:
Just FYI i did search "remote" and it didnt show anything but didnt think to check closed tickets, since its still an issue. but seen the updates, ill check the config file. Thanks.

exelban, 2025-05-04:
please at least use the latest version before oppening an issue. This windows was removed a few version before

scottnzuk, 2025-05-04:
ah i have auto update on but must not rolled around on mac yet. ill force update now.