exelban/stats #2542: Disk space usage info is not correct

booxter, 2025-05-05:
**Describe the bug**

I see different Free / Used numbers in Disk Utility and in the Disk widget of the applet.

[image]

[image]

[image]

```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/disk3s1s1  927G  722G  206G  78% /
/dev/disk3s6    927G  722G  206G  78% /System/Volumes/VM
/dev/disk3s2    927G  722G  206G  78% /System/Volumes/Preboot
/dev/disk3s4    927G  722G  206G  78% /System/Volumes/Update
/dev/disk1s2    500M   20M  481M   4% /System/Volumes/xarts
/dev/disk1s1    500M   20M  481M   4% /System/Volumes/iSCPreboot
/dev/disk1s3    500M   20M  481M   4% /System/Volumes/Hardware
/dev/disk3s7    927G  722G  206G  78% /nix

$ diskutil list
/dev/disk0 (internal, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *1.0 TB     disk0
   1:             Apple_APFS_ISC Container disk1         524.3 MB   disk0s1
   2:                 Apple_APFS Container disk3         994.7 GB   disk0s2
   3:        Apple_APFS_Recovery Container disk2         5.4 GB     disk0s3

/dev/disk3 (synthesized):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      APFS Container Scheme -                      +994.7 GB   disk3
                                 Physical Store disk0s2
   1:                APFS Volume Macintosh HD            11.2 GB    disk3s1
   2:              APFS Snapshot com.apple.os.update-... 11.2 GB    disk3s1s1
   3:                APFS Volume Preboot                 7.1 GB     disk3s2
   4:                APFS Volume Recovery                1.0 GB     disk3s3
   5:                APFS Volume Data                    601.1 GB   disk3s5
   6:                APFS Volume VM                      17.2 GB    disk3s6
   7:                APFS Volume Nix Store               136.5 GB   disk3s7
```

**Details:**
 - Device: MacBook M3 Pro
 - macOS: 15.4.1 aarch64
 - Application version: 2.11.42

exelban, 2025-05-06:
please try release v2.11.41

booxter, 2025-05-06:
I tried .41 and the behavior was the same. Here's the comment where I posted screenshots from .41: https://github.com/exelban/stats/issues/2531#issuecomment-2849332138

exelban, 2025-05-07:
So, as I said in the same thread: there is nothing I can do with that. MacOS returns this value and I just visualize that.

booxter, 2025-05-07:
I believe there's a solution that would guarantee you show what the actual disc space use is, as described in this comment: https://github.com/exelban/stats/issues/2531#issuecomment-2850069482 I guess you have your reasons to not pull the data using the CLI tool, but this seems unfortunate that the disk space widget is misleading.