exelban/stats #2545: Network status sometimes reported incorrectly reported as DOWN

uhkeller, 2025-05-08:
**Describe the bug**

When connecting to tethered WiFi provided by my Android phone, the network status is often incorrectly reported as DOWN even though the connection is working fine.

[image]

Clicking the reset button next to "Details" has no effect. It never changes to UP. When I restart Stats, however, it'll work fine from then on. I'm typically using this connection on train rides so sometimes there will in fact be no connectivity, but after the restarts Stats reliably detects the network coming back up.

[image]

I'm using TripMode to filter network usage when connected to the phone, but Stats is on the allow list and a Stats restart fixes it, so it seems unlikely that TripMode is to blame here.

I noticed that in the screenshots the order of the public IP addresses changes after the restart (before IPv6 is first, after IPv4 is first), so maybe this is some kind of IPv6-related problem?

If there is anything I can do to provide more information please let me know.

**Details:**
 - Device: M1 MacBook Pro 2021
 - macOS: 15.4.1
 - Application version: 2.11.42

uhkeller, 2025-07-24:
Since reporting this I've seen it happen on multiple different networks, not only on the (high latency) phone connection. Just now it happened on my workplace's wired network with latency below 3ms. Just to rule out high network latency as a possible trigger.

j01am, 2025-11-18:
Same behavior for me :)

uhkeller, 2026-02-03:
Out of curiosity I asked GitHub Copilot about this. It gave me a plausible-sounding [explanation](https://github.com/copilot/share/427000aa-41a0-8cb1-9840-8402c4834856) (which might still be utter nonsense of course).
Here's the important part:
> The status should ideally use `usage.interface?.status` (the actual interface UP/DOWN state) instead of `reachability.isReachable`, or combine both checks. The interface status is updated on every read cycle and reflects the true state of the network interface, not just internet reachability.