exelban/stats #2548: [Feature Request] Stats modules should be configurable for each connect display

harmandsingh, 2025-05-11:
Background/Context - Stats modules should be configurable independently for each connect display instead of same modules on each display.

Often times my Macbook is connected to an external display and because of limited menu bar space on my macbook with the notch, I would like to enable only 1-2 modules on the built-in mac display while enabling more modules on the connected external display.

exelban, 2025-05-15:
if it's two independent displays that could be configured and works at the same time: just move widgets you want to have to the screen you want to have

closed 2025-05-15 by exelban

harmandsingh, 2025-05-16:
@exelban But it doesn't work like that, I'm talking about two displays connected to the same system. Unless there's a way to do it, if there is please point me to it.