exelban/stats #2550: The app quits every day after a few hours

FlorinAndrei, 2025-05-12:
MacBook Pro with macOS 15.4.1. I have the latest app version installed (2.11.42).

The app works fine, except it inevitably quits after several hours of running. It just disappears suddenly. I have to start it again manually.

This happens pretty much every day.

This is a development machine, I have Docker Desktop, I run IDEs such as VSCode, etc. The machine is rarely starved for resources. CPU usage is usually less than 100%, and RAM usage is typically around 50%.

exelban, 2025-05-13:
Hi. Please share the crash report

FlorinAndrei, 2025-05-13:
I've pasted the most recent crash here, it's the whole report from the Console:

https://gist.github.com/FlorinAndrei/f68d23816289a6de60e005a36778a7e5

Let me know if you need anything else. Thanks!

exelban, 2025-05-13:
@FlorinAndrei thx. Please try to disable Network module and check if the app will still crash

FlorinAndrei, 2025-05-17:
@exelban I have disabled the Network modules 3 days ago. Since then, there was no crash whatsoever. Usually, the app would crash at least once a day. Looks like the cause is related to that module.

exelban, 2025-05-17:
@FlorinAndrei thx, what widgets have you enabled in the Network module?

FlorinAndrei, 2025-05-19:
@exelban Is it enough if I just give you the entire settings file? See below.

[Stats.plist.txt](https://github.com/user-attachments/files/20300942/Stats.plist.txt)

Let me know if you need anything else.

FlorinAndrei, 2025-05-20:
I've re-enabled the Network module because I need it (it's the one I use most often) and, sure enough, the app has crashed today.

FlorinAndrei, 2025-05-22:
I've installed this new build that you gave me, and the app crashed again.

Crash report: https://gist.github.com/FlorinAndrei/b0eff9c719bdbf1fb6852c92e770a456

exelban, 2025-05-23:
hmm, could you please disable the line chart widget for network module and enable some another widget for this module?

FlorinAndrei, 2025-05-30:
I started using that special build since you posted it here, and there are no more crashes, even though the Network graph is enabled. Before, the app was crashing daily.

Looks like the bug is fixed in the special build.