exelban/stats #2553: Menu Bar Latency Notifications

WillFulmer, 2025-05-14:
Using a quite old application called autoping, I could visualize ping response time and packet loss within my MacOS Menu Bar.

Is there any way to do this using Stats?

Very basic, but provides easy visual clarity:

[image]

[image]

exelban, 2025-05-14:
yes, you can use Text widget for that. Just enable it and set a needed value for that.

closed 2025-05-14 by exelban

WillFulmer, 2025-05-14:
Ah that's excellent! Thank you
I have merged the widgets

And am using $icmp.latency

Any way to determine packet loss?

[image]