exelban/stats #2561: Network download data state is missing in menu bar

Giuliao, 2025-05-18:
**Describe the bug**
network download data state is missing in menu bar
[image]

[image]

**Details:**
 - Device: Macbook M3 Pro 
 - macOS: 10.15.5
 - Application version: 2.11.42

closed 2025-05-18 by exelban

Giuliao, 2025-05-19:
Hello~ Can you show me the same opened issue so that I can follow up fixing progress. Thanks @exelban 
> I had tried to search issues by keyword 'network', but I did not find.

douglasjunior, 2025-07-01:
Just to help anyone looking for a solution:

The solution is to switch to the Process based type.
See this comment for more details: https://github.com/exelban/stats/issues/2461#issuecomment-2822739996