exelban/stats #2562: [UIX] Improve the width transitions on value change

mghenciu, 2025-05-19:
**Describe the issue**
First of all thank you for contributing to the project 🚀  and secondly - this is more of a suggestion, not a bug (so no rush with it).

The transition between different text values in the Menu Bar causes the container width to change, which can be visually distracting as it shifts other items. From a UX perspective, it would be better if the container maintained a consistent width to avoid this shifting.

For example, with my Hubstaff time tracking app - when the Stats widget is near, the changing values and the resulting width changes cause the trigger to move.

This seems to be related to the varying character widths in the typography. **A potential improvement would be to set a minimum width for these containers.**

Consider a scenario with two sensors displaying values. Even if the displayed number has the same number of digits (e.g., two digits), the container width can still change due to different font widths. Setting a minimum width of, say, 58px (based on the widest two-digit combination) could prevent this.

Here's an example illustrating the issue with different font widths affecting container size even with the same number of digits: [[link to where you've made the example, as mentioned in your original text](https://www.figma.com/design/ZYuCX9SDQngnrBTS1dFiZW/Draft?node-id=0-1)]

Ping me if anything is unclear.

[image]

Ping me in case something is not clear.

**Details:**
 - Device: Macbook Pro M1
 - macOS: 15.5 (24F74)
 - Application version: Version 2.11.43

exelban, 2025-05-20:
@mghenciu hi, thx for the feedback.
Im not sure I understand that. Do you propose to increase the font size or spacing between values?

mghenciu, 2025-05-20:
Hi @exelban,

Sorry if this was too abstract 🙈 
I was suggesting to add a **min-width** for the container.

I made a prototype in [Figma](https://www.figma.com/design/YoD43S4zBxjBNRkS7bjvGS/Stats-Width?node-id=1-159&t=MB8KHl6Kv6z0yLm0-1) so it's easier to understand the issue (adde some color on the Stats widget, so the change is more visible).

Here it's how [it behaves now ](https://www.figma.com/proto/YoD43S4zBxjBNRkS7bjvGS/Stats-Width?page-id=0%3A1&node-id=1-159&viewport=115%2C234%2C0.52&t=2aGRVSItkeb8aG67-9&scaling=min-zoom&content-scaling=fixed&starting-point-node-id=1%3A159&show-proto-sidebar=1)(when there is no min-width set):

------

And here is what [I wanted to suggest](https://www.figma.com/proto/YoD43S4zBxjBNRkS7bjvGS/Stats-Width?page-id=0%3A1&node-id=2-362&viewport=115%2C234%2C0.52&t=2aGRVSItkeb8aG67-9&scaling=min-zoom&content-scaling=fixed&starting-point-node-id=2%3A362&show-proto-sidebar=1) (here there's a min-width):

https://github.com/user-attachments/assets/12bf80ff-873a-4f7e-8e43-688f332bad73

I hope this makes more sense now.

romanr, 2025-06-04:
Maybe you mean that proportional font should be used?

mghenciu, 2025-06-04:
> Maybe you mean that proportional font should be used?

Thanks for your input, Roman.

Probably the opposite of proportional font. 
So if a fix via font font update is the way -> then a Monospaced font should be used, because in that case each character occupies the same horizontal space.

romanr, 2025-06-04:
Oh yes I meant fixed font. @exelban I can make change and PR if you ok with that.

exelban, 2025-06-05:
no, you cannot. I assume by fixed font you mean monospaced font. If yes:
- there is an option to enable that in the widget settings
- it cannot be enabled by default since this font causes crash on some Macs

romanr, 2025-06-05:
I couldn't find the option for mono font, where is it?

exelban, 2025-06-05:
in the widget settings