exelban/stats #2594: another crash

FlorinAndrei, 2025-06-11:
**Describe the bug**
I just had another crash today. This is with the latest app version.

**Details:**
 - Device: MacBook Pro 2023
 - macOS: 15.5
 - Application version: 2.11.45

Stats configuration: https://gist.github.com/FlorinAndrei/61a5847dae90bae5b46c7153ae972cd5

Crash report: https://gist.github.com/FlorinAndrei/634ceb662531d9a04dbcf45f302a0bac

FlorinAndrei, 2025-06-13:
And another crash today:

https://gist.github.com/FlorinAndrei/9cd30787ff3550d667ada9feabaa537e

exelban, 2025-06-14:
hi, please try this build:

[Stats.dmg.zip](https://github.com/user-attachments/files/20738612/Stats.dmg.zip)

FlorinAndrei, 2025-07-12:
It took me a while to systematically try each module, but finally I did it. Here are the results:

I got crashes, and quite often (about once a day, or even twice a day), when disabling any of these modules:

- CPU
- GPU
- RAM
- Disk

I've disabled the Network module several days ago (4...5 days maybe), and I have not seen a crash so far. I wish I could continue the experiment a while longer, to make it more relevant, but I will not use this laptop for the next 2 weeks, and I will not be able to respond to this ticket until after 2 weeks.

It seems like disabling the Network module prevents the crashes. The confidence level is not super high.

I don't know if you've tried this, but I have Docker Desktop installed here, I run some containers occasionally, I build Docker images, etc. Docker has a local network that it creates. It also goes to sleep after some minutes. I don't know if Stats gets in trouble because of the Docker network changes, but it might be worth trying.

I will look again at this ticket in about 2 weeks.

Cheers!

FlorinAndrei, 2025-08-02:
I came back from vacation 1 week ago. This whole week (plus 5 days before vacation) Stats has run with the Network module disabled. Not a single crash this whole time.

I am going to re-enable Network now, since I need it anyway.

exelban, 2025-08-02:
Hi, thx. 
Ok, so looks like still something in the network. The last time there was a line chart widget. Could you check if it's still this widget causes crash by disabling it for network?

FlorinAndrei, 2025-08-04:
Okay, I kept the Network module enabled, but I moved the chart widget to the right of the vertical bar in settings, disabling it. I only see NET now but I don't see any chart there. I hope that's what you want.

https://gist.github.com/FlorinAndrei/faec9b8cfffe8a50b9d10eac98e3662b

FlorinAndrei, 2025-08-07:
It crashed today. The Network module enabled, but the chart widget was disabled. It looks like it's not the chart, but maybe the module itself.

I will re-enable the chart now.

Anything else to try?

FlorinAndrei, 2025-08-14:
No crash in 1 week with v2.11.49

I've re-enabled the Network chart widget.

FlorinAndrei, 2025-08-14:
I take that back. There was a crash in the past 7 days. In the Console, I've noticed a crash from 2 days ago, with v2.11.49 and the Network chart widget disabled. Here is the report:

https://gist.github.com/FlorinAndrei/8b3f7715a4a1ae9d20c5f10a50ff259d

And then today, pretty soon after enabling the Network chart widget, it crashed again. Here is the report:

https://gist.github.com/FlorinAndrei/60b59a1a5b31800d6b3ebd9422caf9d5

I will keep the Network chart widget enabled, because I need it.

If you want more data, maybe create a build with debugging turned on, or something?

FlorinAndrei, 2025-08-15:
> change the charts history to some other value

Is that the setting called "Number of reads in the chart"?

FlorinAndrei, 2025-08-16:
crash report: https://gist.github.com/FlorinAndrei/3d253fcbff4c38f4deccde401115d9ce

Stats config: https://gist.github.com/FlorinAndrei/7cdfd6246b03de4be8dafc77e9f9c5f4

FlorinAndrei, 2025-08-20:
using the latest special build

crash report: https://gist.github.com/FlorinAndrei/45d8c59ddc794bdde906967a25e9a23f

stats config: https://gist.github.com/FlorinAndrei/f58baead239608f371685ec18a08667f

exelban, 2025-09-01:
please ping me in case someone will have the same issue and could run Stats in Xcode to debug that.