exelban/stats #2603: Fan control doesn't work with the new M4 Mac Studio's

JustJoostNL, 2025-06-18:
**Describe the bug**
The fan control feature does not work with the new M4 Mac Studio's. The `macs-fan-control` app had the same issue, and that seems to be solved, see: https://github.com/crystalidea/macs-fan-control/issues/817 and https://github.com/crystalidea/macs-fan-control/issues/823

Would it be possible to add support for this too? I guess you can look at the code of `macs-fan-control` to check how to implement this.

**Details:**
 - Device: Mac Studio 2025
 - macOS: 15.5
 - Application version: 2.11.45

exelban, 2025-06-18:
there is no code of macs-fan-control

closed 2025-06-18 by exelban

JustJoostNL, 2025-06-18:
@exelban My bad. But why does that make it not planned?

exelban, 2025-06-18:
Unfortunately I have no access to the m4 or m3 platform. So I cannot figured out why current approach does not work. 
Also, take a look at FAQ please