exelban/stats #2643: Public IP changed notification appears, when IP is unchanged

PuzzledUser, 2025-07-21:
Related to https://github.com/exelban/stats/issues/2637 (which would make spotting this easier), I noted my public IP half an hour ago (as reported by Stats) and cleared the previous change notification. 

Now I've just seen another notification that my public IP has changed. The Stats reported IP has not changed. 

NB: I am using iCloud Private Relay which rotates the IP seen by Safari websites. However, the UI of Stats correctly reports the unchanged IP provided by my ISP (not one seen by websites).

**Details:**
 - Device: Mac min M2
 - macOS: 15.5 (24F74)
 - Application version: Version 2.11.47

PuzzledUser, 2025-08-26:
I installed Version 2.11.50. However, after a day sleep, then wake I've just seen a notification of a changed Public IP (IP has not changed and is reported correctly). Will keep you informed of any recurrence on Version 2.11.51.

exelban, 2025-08-26:
emm, most probably it's because you IP was changed?

PuzzledUser, 2025-09-25:
Public IP reported as changed today after wake. Confirmed reported IP remains unchanged.
Using v2.11.52. Will update to .53 now and monitor.

PuzzledUser, 2025-10-16:
Stats Version 2.11.57 still sometimes reports Public IP changed (often on wake) when the IP has not changed. https://github.com/exelban/stats/issues/2637 doesn't help clarify the situation.

PuzzledUser, 2025-10-29:
Another notification of Public IP changed today. IP has not changed. https://github.com/exelban/stats/issues/2637 is no help as it just reports the current IP, not what it thinks the previous IP was! 

Was on 2.11.57 will update to .58 and monitor.

PuzzledUser, 2025-11-04:
Currently, I have the benefit of two swappable public IPs so I can test. I can see that when I manually refresh the IP, the notification now shows **both** previous and new IP. 
However, this suggests that the bug could be that Stats has no record of the previous IP (after wake) sometimes. Hence it doesn't display it and sees the presence of an IP as a change from _nothingness_.