exelban/stats #2694: Suddenly no stats showing up on the menubar

IsmailAlamKhan, 2025-09-04:
I have been using Stats for a long time almost 2 years, it has been working flawlessly up until now. I was playing with the Settings and customizations, I disabled the "Combined Modules" setting to see what it does now no stats are showing up on my menubar

exelban, 2025-09-04:
hi, you can reset the settings with `defaults delete eu.exelban.Stats`.

IsmailAlamKhan, 2025-09-04:
Yes it does

exelban, 2025-09-05:
so or you have disabled modules, or there is no space in the menu bar

IsmailAlamKhan, 2025-09-05:
[image]

I have more then enough space

[image]

[image]

IsmailAlamKhan, 2025-09-06:
@exelban can you please reopen the issue? As this is still not fixed.

IsmailAlamKhan, 2025-09-08:
@exelban Hey are you not going to fix my issue?