exelban/stats #2704: Nothing shows up in the menu bar on macOS 26

Parakeet709, 2025-09-16:
I just upgraded to macOS 26 and when I check in the menu bar - boom, nothing. None of my temperatures, battery, RAM monitor, network monitor, nor my clock were visible in the menu bar. I checked in the system settings app, and all the permissions were correctly configured. I was also able to use the app just fine on macOS 15. 

Version 2.11.52
macOS Tahoe 26.0.0

jrsacher, 2025-09-16:
I noticed this and played around a bit. I found that there's an "Allow in the Menu Bar" setting that needs to be enabled. Give that a shot!

[image]

duncangeere, 2025-09-16:
I have the menubar permissions enabled, and I see my network status, but the battery status is missing, even after I toggle it on and off in stats. Any idea what might be going on?

exelban, 2025-09-16:
I have updated 2 macs to macOS 26, and on both, everything works fine out of the box. I have read a few threads and it looks like there was a bug in the betas related to this. Maybe one of you was using the beta before?

duncangeere, 2025-09-16:
Apologies - I think this was user error on my side. What seems to have happened is that the order of menubar items shifted, and a bunch of things (like my battery status) got hidden behind the notch. Confirming I'm seeing both network and battery status now, on version 2.11.52.