exelban/stats #2733: Widgets do not update or high disk writes

exelban, 2025-09-30:
Hi, in the new macOS 26 there is an internal bug with the internal process `chronod`. As I understand this process is used in communication between the app and widgets. So each time Stats has a new value to show it triggers the widget updates. There is nothing that I can do from my side with that for now. And the best I can do at this moment is disable widget updates for the new macOS 26. I will keep an eye on this situation, but at this moment I have no solution. I was waiting for the first patch from Apple with the hope that it would fix the problem, but since nothing changed, I released a hot fix that disables the widget updates.

This issue is created to prevent duplicates.

streamofstars, 2025-12-10:
I just upgraded my macOS to Tahoe 26.1 and right away noticed the thing with Stats constantly writing to disk via chronod process. Restarting macOS does not help. Restarting Stats does not help. In the post above you suggest installing specific build which is v2.11.60 but I already have 2.11.62 installed. Should I downgrade?

artursudnik, 2025-12-11:
> I have add an option to disable widgets updates, it's available in the app settings.

what actually changes after I disable this?

naokihoshino, 2025-12-25:
Please set the toggle "off" by default. There is a chance most of them have not noticed the disk writes.

mksoska, 2026-08-24:
A similar issue regarding chronod disk writes was observed and fixed here: https://github.com/steipete/CodexBar/issues/2369
@exelban maybe you could take a look a verify whether the patch is applicable to Stats